GraffitiResearchLabGermany/bombIR #116: Relicense under GPL

aalex, 2016-04-05:
Hi,
CC-By-SA is discouraged for licensing source code. http://opensource.stackexchange.com/questions/1717/why-is-cc-by-sa-discouraged-for-code

Would you like to relicense this prototype under GPL? I had to create a similar graffiti application, and I took your code as inspiration. I would gladly refer to you guys as co-authors and original coders. The only problem is that CC-By-SA is not well-suited for source code licensing. Would you please allow me to state that this code has been relicensed under the GPL?

Thank you.
Best regards, Alexandre
http://www.sourcelibre.net

JesseScott, 2016-04-06:
@SableRaf @tugelbend thoughts ?

tugelbend, 2016-04-07:
Sure!

JesseScott, 2016-04-08:
@aalex why dont you submit a PR ? :dancer: 
also, jesus, just saw your last name and realized that it was you! Hi!!! :wave: